sombriks/node-libgpiod #10: input debounce?

mhelmarc, 2023-01-25:
Hi, first off, I'm trying to learn nodejs and gpio, and your module is the most platform independent I could find.
Almost all of gpio module for nodejs is geared toward raspberry pi, I'm on an Alwinner (nanopi neo) SOC.
The module works as demoed, but I'm using buttons and sometimes I'm getting more than one result on a single press,
Do you know of a workaround for it?

noctarius, 2023-01-25:
This has nothing to do with the library but is a very common occurrence with push buttons where the electrical signal is picked up multiple times due to mechanical reasons (commonly referred to as bouncing). You can either fix this in hardware (https://www.digikey.de/en/articles/how-to-implement-hardware-debounce-for-switches-and-relays) or software (https://docs.arduino.cc/built-in-examples/digital/Debounce).

mhelmarc, 2023-01-25:
Thanks, I was hoping for something similar with the 'onoff' module.

noctarius, 2023-01-25:
This library only wraps the libgpiod with a nodejs api. Debouncing isn't built in but it is easy enough to do. Feel free to provide a PR to add it as an additional module. Something like passing in a Line instance into a PushButton instance (or something like it) and have it available for all people in the future :)

mhelmarc, 2023-01-26:
If I ever get proficient with js I will :-)
Thanks.

adminy, 2023-07-28:
I'm guessing another solution to this would be the monitoring callbacks but that's not yet implemented, but the idea is that it monitors any changes on the input pins selected and logs an event of raise or fall of an edge.

sombriks, 2024-02-25:
closing but hopefully will get solved when monitoring finally gets implemented. see #23

closed 2024-02-25 by sombriks