sombriks/node-libgpiod #11: Missing gpiod_line_request_input_flags

splitice, 2023-02-15:
Without support for gpiod_line_request_input_flags (which perhaps should be an additional argument in the requestInputMode) its not possible to set a pull up or pull down on the line.

This is the only thing preventing this library from being almost a drop-in replacement (that I have noticed) for the capabilities offered by most sysfs based APIs.

d-beamon, 2024-02-20:
Any plans for this?

sombriks, 2024-02-25:
feature should be available on 0.4.0 Release, anyone able to test it please provide feedback. here.

sombriks, 2024-02-25:
https://github.com/sombriks/node-libgpiod/pull/21

splitice, 2024-02-25:
Nice! I can certainly test it this week.

splitice, 2024-03-05:
From a usability / API perspective I would have that method called "requestInputFlags" as it does not set the mode of the GPIO as `requestInputMode` as does.

Other than that it seems to work as designed.

sombriks, 2024-03-05:
> From a usability / API perspective I would have that method called "requestInputFlags" as it does not set the mode of the GPIO as `requestInputMode` as does.
> 
> Other than that it seems to work as designed.

That's nice!

we can enhance the API design in the upcoming 1.x version. it's done for now.

closed 2024-03-05 by sombriks