sombriks/node-libgpiod #19: Suppport for Raspberry Pi 5

RickBullotta, 2024-01-28:
Any plans to support the Rpi 5?

sombriks, 2024-01-28:
Hello @RickBullotta !

Technically speaking yes, if RPI-5 runs any linux offering libgpiod then it's supposed to be supported.

Main issue at the moment is a recent breaking change [inside libgpiod itself](https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/refs/); we're planning to branch and sync versions accordingly so 1.x and 2.x branches will get along libgpiod versions.

sombriks, 2024-02-25:
anyone able to run any of [hello-world](https://github.com/sombriks/node-libgpiod-examples) projects on RPI5 so we can confirm that it 'just works' besides our lack of complete features?

alessandromrc, 2024-03-24:
Can confirm that [Hello-world](https://github.com/sombriks/node-libgpiod-examples) runs pretty good, also removing the timeouts and the counting like this:

```js
const { version, Chip, Line } = require("node-libgpiod");

global.chip = new Chip(4);
global.line = new Line(chip, 17); // led on GPIO17
let count = 10;

console.log(version());
line.requestOutputMode();

let value_out = false;

while (true)
{
  line.setValue(value_out);
  value_out = !value_out;
}
```

Pulls out around 1MHz which is surely not bad for NodeJS.

sombriks, 2024-03-25:
@alessandromrc that's great!

As soon as i finish the 1.x and 2.x branches for the two versions of libgpiod i ca turn my eyes back to performance, maybe squeeze a little more!

closed 2024-03-25 by sombriks

alessandromrc, 2024-03-25:
> @alessandromrc that's great!
> 
> As soon as i finish the 1.x and 2.x branches for the two versions of libgpiod i ca turn my eyes back to performance, maybe squeeze a little more!

One way I could think of is delegating tasks directly to the underlaying library for stuff like clock generation etc...

dmitrydvorkin, 2024-03-27:
> One way I could think of is delegating tasks directly to the underlaying library for stuff like clock generation etc...

it is traditionally done by the kernel driver, not by the user-space library (which should be compatible with non-rpi linux world)