sombriks/node-libgpiod #25: Required version of libgpiod?

splitice, 2024-03-06:
Potential min libgpiod version or ifdefs required.

```
../src/line.cc:24:99: error: ‘GPIOD_LINE_REQUEST_FLAG_BIAS_DISABLE’ was not declared in this scope; did you mean ‘GPIOD_LINE_REQUEST_FLAG_OPEN_DRAIN’?
   24 |   Nan::Set(lineFlags, Nan::New("GPIOD_LINE_REQUEST_FLAG_BIAS_DISABLE").ToLocalChecked(), Nan::New(GPIOD_LINE_REQUEST_FLAG_BIAS_DISABLE));
      |                                                                                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                                                   GPIOD_LINE_REQUEST_FLAG_OPEN_DRAIN
../src/line.cc:25:101: error: ‘GPIOD_LINE_REQUEST_FLAG_BIAS_PULL_DOWN’ was not declared in this scope; did you mean ‘GPIOD_LINE_REQUEST_FLAG_ACTIVE_LOW’?
   25 |   Nan::Set(lineFlags, Nan::New("GPIOD_LINE_REQUEST_FLAG_BIAS_PULL_DOWN").ToLocalChecked(), Nan::New(GPIOD_LINE_REQUEST_FLAG_BIAS_PULL_DOWN));
      |                                                                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                                                     GPIOD_LINE_REQUEST_FLAG_ACTIVE_LOW
../src/line.cc:26:99: error: ‘GPIOD_LINE_REQUEST_FLAG_BIAS_PULL_UP’ was not declared in this scope; did you mean ‘GPIOD_LINE_REQUEST_FLAG_ACTIVE_LOW’?
   26 |   Nan::Set(lineFlags, Nan::New("GPIOD_LINE_REQUEST_FLAG_BIAS_PULL_UP").ToLocalChecked(), Nan::New(GPIOD_LINE_REQUEST_FLAG_BIAS_PULL_UP));
      |                                                                                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                                                   GPIOD_LINE_REQUEST_FLAG_ACTIVE_LOW
```

Havent yet dug into libgpiod to work out the required version.

sombriks, 2024-03-06:
Hi @splitice !
Indeed there is a new version of libgpiod in streets and we intend to maintain both.

in this repo there is [main-1.x branch](https://github.com/sombriks/node-libgpiod/tree/main-1x) which is supposed to work with libgpiod 1.x series and [main-2.x branch](https://github.com/sombriks/node-libgpiod/tree/main-2x) for the new version.

the 2.x series will introduce breaking changes because libgpiod itself changed, and we'll try to mirror the original api as much as possible.

please check which libgpiod version your system have; if your kernel is updated it is most likely you will need to wait for a 2.x release

splitice, 2024-03-28:
It might be best to use ifdefs to make a dual compatible version?

From a project maintenance point of view its a metric PITA for our developers with a different version of libgpiod to the production system during development. And I've checked debian doesnt have a stable release with libgpiod 2.x yet.

Even ifdef on `GPIOD_LINE_REQUEST_FLAG_BIAS_PULL_UP` for 1.x that should be fine?

splitice, 2024-03-28:
I got to say ubuntus versioning on libgpiod is very confusing.

I'm on 20.04 (LTS) for development.
```
Package: libgpiod-dev
Architecture: amd64
Version: 1.4.1-4
[...]
Depends: libgpiod2 (= 1.4.1-4)
Suggests: libgpiod-doc
Filename: pool/universe/libg/libgpiod/libgpiod-dev_1.4.1-4_amd64.deb
```

I thought you were right initially but instead it looks like the version is 1.4, perhaps 1.4 is too old?

Those defines were added in 2019 and are not in v1.4.0. I would confirm this against the source repository at kernel.org, however I'm getting 403's currently. I confirmed it however against a github mirror.

This means that we perhaps need either to ifdef these exposed defines, or raise the documented supported version in the README

splitice, 2024-03-28:
Can confirm v1.6.3 (package from ubuntu 22.04) contains the required definitions.

sombriks, 2024-03-28:
just to be clear, the definitions are missing from index.d.ts in 1.x branch, is that correct?

splitice, 2024-03-30:
Huh? No.

You either need some ifdefs to support v1.4, or to add a min requirement in the README for v1.5.

sombriks, 2024-04-06:
Hi @splitice !
It take a while but i managed to try something to solve it.

The first attempt is indeed to add a compile flag so i can proper setup it in the future.

If possible, please check if it works again with older libgpiod (prior to 1.5) and let me know!

splitice, 2024-04-09:
`TypeError: Cannot read properties of undefined (reading 'GPIOD_LINE_REQUEST_FLAG_ACTIVE_LOW')`

This is on debian bookworm `libgpiod2 1.6.3` which does have those flags

sombriks, 2024-04-09:
Hey @splitice , thanks for the feedback, looks like it's time to debian virtual machine then! i'll figure out what's the issue.

sombriks, 2024-06-30:
Hello @splitice ,

i moved the enums definition to javascript side of the code; i hope this change makes transparent for the node binding if it's being built against 1.4, 1.5 or 1.6 libgpiod version.

when you have a chance please run some tests using the 0.4.5 version from npm, or even checking test cases against gpio-sim setup (not perfect yet, but try it out if your environment sits in the sweet spot of having the gpio-sim kernel module and libgpiod version prior to 2.0).

sombriks, 2025-05-18:
i'll give up on older versions since i need to focus on 2.x soon.
we can discuss libgpio 1.4 before api freeze for 1.x

commit 64fda910 (2025-05-18): closes #25
closed 2025-05-18 by sombriks