sombriks/node-libgpiod #27: No longer compatible with libgpiod < 1.6.x

alejandroclaro, 2024-04-05:
The documentation states in the requirements that this should be compatible with libgpiod 1.x. However, since version 0.4 it's no longer compatible with version lower than 1.6.0

[image]

It would be nice its compatibility could be maintained since LTS version of many distros, like Ubuntu 20.04, are still ship with libgpiod 1.4.

If not possible, I recommend updating the documentation and clarify the minimum requirements.

sombriks, 2024-04-06:
Hello @alejandroclaro , thank you for reporting this!
I think it is related to what @splitice was warning me about in #25 and indeed, i should add some kind of ifdef or something to make it possible to build node-libgpiod again  under older versions -- as long as it is 1.x

sombriks, 2024-04-06:
Hello @alejandroclaro when you have a chance, please check if it works again with libgpiod versions prior to 1.5 series.

alejandroclaro, 2024-04-09:
Thanks @sombriks !

I will test this today.

alejandroclaro, 2024-04-09:
@sombriks  looks like everything is good now.

Thanks!

closed 2024-04-09 by alejandroclaro