sombriks/node-libgpiod #30: When importing mysql2 and setting the value of lines in a setInterval it gives an EPERM error

LaurentDhont, 2024-05-28:
This is a very strange issue, see below a piece of code that works, but when you `require('mysql2') `it stops working and gives the following error: 
```
Error: EPERM, Operation not permitted
    at Timeout.main [as _onTimeout] (test.js:9:7)
    at listOnTimeout (node:internal/timers:573:17)
    at process.processTimers (node:internal/timers:514:7) {
  errno: 1,
  code: 'EPERM',
  syscall: '::setValue'
}
```
Code:
```
const lib = require('node-libgpiod');
const mysql2 = require('mysql2'); // when commenting this line everything works fine
const chip = new lib.Chip(4);
const line = new lib.Line(chip, 26);
line.requestOutputMode(0);

function main() {
        line.setValue(1);
}

setTimeout(main, 1000 * 30);
```
Versions:
Node: v20.13.1
mysql2: 3.9.8
node-libgpiod: 0.4.3
ubuntu: Ubuntu 24.04 LTS
Raspberry Pi 5B rev 1.0

sombriks, 2024-06-01:
Hello @LaurentDhont 
Thanks for report this, i'll investigate as soon as i get the work on gpiosim done.

sombriks, 2024-06-30:
Hello @LaurentDhont , 

indeed an odd issue, you can help me out to debug it by cloning the examples repository and running this one: https://github.com/sombriks/node-libgpiod-examples/tree/main/database-read-pin-write

i am still configuring other real hardware environments, but i was unable to reproduce the issue on my virtual machine or in my raspberry pi zero w (32 bits running raspbian 12 / node 18)

i still have a rasberry pi model 3 B+ to test and my new board, a raxda rock 3C, but i don't have a raspberry pi 5 at hand.

let me know when you have a moment to spare and try the sample code i built to explore this issue.

sombriks, 2025-05-18:
tested the sample project against latest release and it still works.