sombriks/node-libgpiod #38: GPIO pins 2 - 8 work great but 9 and up not

bengoevaerts, 2025-02-24:
I wonder what I'm missing with the following issue:
I'm testing all pins to connect multiple push buttons to the PI. My code is working great for pins 2 till 8 but starting from pin 9 (and all pins following), `line.getValue()` randomly seems to return 0 and 1.

In Python reading (for example) pin 9 is working as expected.
I'm on a Raspberry PI 5 and using `global.chip = new Chip(4);`

Any help or direction is appreciated. Thanks.

noctarius, 2025-02-24:
For stability, I'd recommend using external pull-up resistors. Internal pull-ups might not be strong enough for stable operation. 

In any case, it seems like on a Raspi, only the first 8 pull-ups are enabled by default on bootup. The other ones need to be enabled manually. Not sure if this is still true, as the answer is from 2018: https://forums.raspberrypi.com/viewtopic.php?t=202568#p1259134

> When a Pi is first powered up, the first eight GPIOs have pull-ups enabled, and the rest have pull-downs enabled. After it has finished booting, you can change that if you wish, in your software.

bengoevaerts, 2025-02-25:
Thanks for the quick reply. The blog post does not mention how to enable pull-up mode. Based on your comment I assume it's not possible using the `[node-libgpiod](https://github.com/sombriks/node-libgpiod)` library which would be useful!

sombriks, 2025-05-26:
hi everyone!

can someone test release 0.5 ? it is supposed to offer a way to configure bias, but i lack the hardware to do so at the moment.